sdr-enthusiasts/docker-flightradar24 #5: raspberry pi 4 (linux/arm64/v8) support

emilburzo, 2020-05-09:
Hi,

First off, thanks for building this (and `piaware`), they are setup exactly as I would imagined they should be and make transitioning from a manual setup to a docker based one incredibly easy.

Not sure how difficult it is, but would it be possible to have a `fr24feed` docker image compatible with raspberry pi 4?

```
Unable to find image 'mikenye/fr24feed:latest' locally
latest: Pulling from mikenye/fr24feed
docker: no matching manifest for linux/arm64/v8 in the manifest list entries.
```

mikenye, 2020-05-11:
Hi @emilburzo, if you could please re-pull `mikenye/fr24feed:latest` and give it a try, I've included support for `arm64` architecture.

A note on `arm64`: FlightRadar24 only make binaries available for `amd64` and `armhf`. The `arm64` version of this container uses the `armhf` binaries which are compiled for `arm32`. `arm32` support is optional on `arm64`. In practice, there is only one `arm64` CPU that omits legacy `arm32` instruction set support - Cavium ThunderX. Thus, this image should work on any `arm64` system that doesn't have the Cavium ThunderX CPU. [Reference](https://askubuntu.com/questions/928249/how-to-run-armhf-executables-on-an-arm64-system).

Given you're using a Pi 4, you should be fine. I've tested the image on my Pi 4 here running `arm64` and it appears to work fine.

emilburzo, 2020-05-11:
@mikenye wow, that was incredibly fast!

I just tried it and everything seems ok. There's little traffic at the moment in my area, so I'll have to wait until there is actually something to track.

The `amd64` and `armhf` is also super interesting and good to know for the future.

I think we can close this issue and I'll follow-up if anything seems weird later on.

Thank you again! this saved me a lot of manual setup

mikenye, 2020-05-11:
No problems!

closed 2020-05-11 by mikenye