sdr-enthusiasts/docker-flightradar24 #14: very high CPU usage

klara31, 2020-09-15:
What happened since the last update, fr24feed is not feeding anymore and CPU is skyhigh?

mikenye, 2020-09-15:
See: https://github.com/mikenye/docker-flightradar24#notice

> RPi users are reporting issues with the latest version (`1.0.26-4`) of the `fr24feed` binary from FR24. RPi users are encouraged to remain on version `1.0.25-3` until these issues are resolved. To do this, use the image `mikenye/fr24feed:1.0.25-3_armhf`. Thanks.

:-)

mikenye, 2020-09-15:
I've hard coded version `1.0.25-3` into the build process for the time being and am running a build. This will revert `latest` to `1.0.25-3` and should fix everyone's issues until FR24 sort themselves out.

FR24 have released `1.0.26-5` and it's still very broken...

mikenye, 2020-09-15:
OK, builds are complete and broken tags removed from docker hub. `latest` tags will now use `1.0.25-3` and will stay that way until FR24 sort out the bugs.

klara31, 2020-09-15:
Just pulled lmikenye/fr24feed:latest again. It still has the issue... reverting back to mikenye/fr24feed:1.0.25-3_armhf

closed 2021-10-06 by mikenye