sdr-enthusiasts/docker-flightradar24 #15: Flight Radar Activation not detecting device

rdelfin, 2020-11-17:
I currently have this container running on my local network. However, I've been unable to get flightradar24 to detect the device in https://www.flightradar24.com/activate-raspberry-pi. Looking at the logs, it's connecting to flightradar24 fine:
```
$ docker logs -f fr24feed
[...]
2020-11-17 22:10:40 | info | [httpd]Server started, listening on 0.0.0.0:8754
2020-11-17 22:10:40 | [i]PacketSenderConfiguration::fetch_config(): Yoda configuration for this receiver is disabled
2020-11-17 22:10:40 | [d]TLSConnection::ctor(): Enable verify_peer in production code!
2020-11-17 22:10:40 | [feed][d]fetching configuration
2020-11-17 22:10:41 | [feed][c]Max range AIR: 350.0nm
2020-11-17 22:10:41 | [feed][c]Max range GND: 100.0nm
2020-11-17 22:10:41 | [feed][c]Timestamps: optional
2020-11-17 22:10:41 | info | [stats]Stats thread started
2020-11-17 22:10:41 | info | Stopping ReceiverACSender threads for feed 
2020-11-17 22:10:41 | info | Configured ReceiverACSender: 185.218.24.22:8099,185.218.24.23:8099,185.218.24.24:8099, feed: XXXXXXX, send_interval: 5s, max age: 15s, send metadata: true, mode: 1, filtering: true
2020-11-17 22:10:41 | info | Network thread connecting to 185.218.24.22:8099 for feed XXXXXXX
```

I've tripled check the docker-compose.yml file and the FR24KEY env var, and they all look fine. What could be the cause of this, and am I missing anything?

```
version: '2.0'

volumes:
  readsbpb_rrd:
  readsbpb_autogain:

services:
  readsb:
    image: mikenye/readsb-protobuf
    tty: true
    container_name: readsb
    hostname: readsb
    restart: always
    devices:
      - /dev/bus/usb/001/005:/dev/bus/usb/001/005
    ports:
      - 8080:8080
      - 30005:30005
      - 30003:30003
    environment:
      - TZ=Europe/London
      - READSB_DCFILTER=true
      - READSB_DEVICE_TYPE=rtlsdr
      - READSB_FIX=true
      - READSB_GAIN=autogain
      - READSB_LAT=XX.XXXX
      - READSB_LON=XX.XXXX
      - READSB_MODEAC=true
      - READSB_RX_LOCATION_ACCURACY=2
      - READSB_STATS_RANGE=true
      - READSB_NET_ENABLE=true
    volumes:
      - readsbpb_rrd:/run/collectd
      - readsbpb_autogain:/run/autogain

  piaware:
    image: mikenye/piaware:latest
    tty: true
    container_name: piaware
    restart: always
    ports:
      - 8081:80
    environment:
      - TZ=Europe/London
      - LAT=XX.XXXX
      - LONG=XX.XXXX
      - BEASTHOST=readsb
      - FEEDER_ID=XXXXXXXXXX

  fr24feed:
    image: mikenye/fr24feed:1.0.25-3_armhf_arm64
    tty: true
    container_name: fr24feed
    restart: always
    ports:
      - 8754:8754
    environment:
      - TZ=Europe/London
      - BEASTHOST=readsb
      - FR24KEY=XXXXXXXXXXX

```

mikenye, 2020-11-18:
Hi @rdelfin,

If you go to <https://www.flightradar24.com/account/data-sharing>, can you see your site there?

Does the status say "Online"? If not, what does it show?

-Mike

rdelfin, 2020-11-18:
HUH yep! And seems like it's showing up in the global stats/ranking now. Maybe I just needed a bit more patience. Thanks! I'll close the issue out

closed 2020-11-18 by rdelfin

mikenye, 2020-11-18:
No problems! I'm glad its working. :)