sdr-enthusiasts/docker-flightradar24 #21: Build fails on Raspberry Pi Zero W

CalvinSchwartz, 2021-01-17:
On a Raspberry Pi Zero W running Raspbian Buster the container fails to start (see: [mikenyefr24feedlatest.log](https://github.com/mikenye/docker-flightradar24/files/5826247/mikenyefr24feedlatest.log)) and also the Build on the Pi Zero fails to complete (see [build.log](https://github.com/mikenye/docker-flightradar24/files/5826258/build.log)).

```
The build fails because the script can't detect the architecture:
+ echo 'ERROR: Unable to determine architecture or unsupported architecture!'
+ exit 1
ERROR: Unable to determine architecture or unsupported architecture!
```

If I comment line 40&47 of the script "deploy_fr24feed.sh" out and force it to use the armhf architecture the build runs through
 and the container itself runs perfectly fine.

**Therefore I suggest to edit the if-function in line 47 of deploy_fr24feed.sh to properly detect the Raspberry Pi Zero W.**

Content of the variable FILEOUTPUT is:
`FILEOUTPUT='/usr/bin/file: ELF 32-bit LSB shared object, ARM, EABI5 version 1 (SYSV), dynamically linked, interpreter /lib/ld-linux.so.3, for GNU/Linux 3.2.0, BuildID[sha1]=f57b617d0d6cd9d483dcf847b03614809e5cd8a9, stripped'`

mikenye, 2021-01-18:
Hi @CalvinSchwartz,

Can you provide the output of `uname -m` and `dpkg-architecture | grep DEB_HOST_ARCH` from your Raspberry Pi Zero W please?

Looking at the `fr24feed` package manifest (see below), we can see that FR24 only make packages available that are compiled for `i386`, `amd64` and `armhf` (which also works for `arm64` due to backwards compatibility.

My understanding is that the Pi Zero W is ARMv6, which (unless I'm mistaken) requires `armel` architecture, which `fr24feed` don't appear to support.

The output of the commands above will confirm this.

Regardless, I've modified the `deploy_fr24feed.sh` script to be more robust.

---

```
$ curl http://repo.feed.flightradar24.com | \
>   xmlstarlet fo | \
>   grep -oP "^\s*<[Kk]ey>([\w\/\.\-]+)<\/[Kk]ey>" | \
>   cut -d ">" -f 2 | \
>   cut -d "<" -f 1 | \
>   grep -i ".deb"
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 71517    0 71517    0     0  56450      0 --:--:--  0:00:01 --:--:-- 56490
linux_x86_64_binaries/fr24feed_1.0.24-5_amd64.deb
linux_x86_64_binaries/fr24feed_1.0.25-1_amd64.deb
linux_x86_64_binaries/fr24feed_1.0.25-2_amd64.deb
linux_x86_64_binaries/fr24feed_1.0.25-3_amd64.deb
linux_x86_binaries/fr24feed_1.0.18-5_i386.deb
linux_x86_binaries/fr24feed_1.0.24-5_i386.deb
linux_x86_binaries/fr24feed_1.0.25-1_i386.deb
linux_x86_binaries/fr24feed_1.0.25-3_i386.deb
pool/raspberrypi-alpha/f/fr24feed/fr24feed_1.0.19-10_armhf.deb
pool/raspberrypi-alpha/f/fr24feed/fr24feed_1.0.19-9_armhf.deb
pool/raspberrypi-alpha/f/fr24feed/fr24feed_1.0.24-1_armhf.deb
pool/raspberrypi-beta/f/fr24feed/fr24feed_1.0.19-11_armhf.deb
pool/raspberrypi-beta/f/fr24feed/fr24feed_1.0.19-12_armhf.deb
pool/raspberrypi-beta/f/fr24feed/fr24feed_1.0.19-13_armhf.deb
pool/raspberrypi-beta/f/fr24feed/fr24feed_1.0.19-15_armhf.deb
pool/raspberrypi-beta/f/fr24feed/fr24feed_1.0.19-5_armhf.deb
pool/raspberrypi-beta/f/fr24feed/fr24feed_1.0.19-6_armhf.deb
pool/raspberrypi-beta/f/fr24feed/fr24feed_1.0.19-7_armhf.deb
pool/raspberrypi-beta/f/fr24feed/fr24feed_1.0.19-8_armhf.deb
pool/raspberrypi-beta/f/fr24feed/fr24feed_1.0.23-5_armhf.deb
pool/raspberrypi-beta/f/fr24feed/fr24feed_1.0.23-8_armhf.deb
pool/raspberrypi-beta/f/fr24feed/fr24feed_1.0.24-1_armhf.deb
pool/raspberrypi-beta/f/fr24feed/fr24feed_1.0.24-2_armhf.deb
pool/raspberrypi-beta/f/fr24feed/fr24feed_1.0.24-3_armhf.deb
pool/raspberrypi-beta/f/fr24feed/fr24feed_1.0.24-4_armhf.deb
pool/raspberrypi-beta/f/fr24feed/fr24feed_1.0.24-5_armhf.deb
pool/raspberrypi-beta/f/fr24feed/fr24feed_1.0.24-6_armhf.deb
pool/raspberrypi-beta/f/fr24feed/fr24feed_1.0.24-7_armhf.deb
pool/raspberrypi-beta/f/fr24feed/fr24feed_1.0.24-8_armhf.deb
pool/raspberrypi-beta/f/fr24feed/fr24feed_1.0.25-3_armhf.deb
pool/raspberrypi-beta/f/fr24feed/fr24feed_1.0.26-4_armhf.deb
pool/raspberrypi-beta/f/fr24feed/fr24feed_1.0.26-5_armhf.deb
pool/raspberrypi-beta/f/fr24feed/fr24feed_1.0.26-7_armhf.deb
pool/raspberrypi-beta/f/fr24feed/fr24feed_1.0.26-8_armhf.deb
pool/raspberrypi-beta/f/fr24feed/fr24feed_1.0.26-9_armhf.deb
pool/raspberrypi-stable/f/fr24feed/fr24feed_1.0.19-11_armhf.deb
pool/raspberrypi-stable/f/fr24feed/fr24feed_1.0.19-12_armhf.deb
pool/raspberrypi-stable/f/fr24feed/fr24feed_1.0.19-13_armhf.deb
pool/raspberrypi-stable/f/fr24feed/fr24feed_1.0.19-15_armhf.deb
pool/raspberrypi-stable/f/fr24feed/fr24feed_1.0.19-5_armhf.deb
pool/raspberrypi-stable/f/fr24feed/fr24feed_1.0.19-6_armhf.deb
pool/raspberrypi-stable/f/fr24feed/fr24feed_1.0.19-8_armhf.deb
pool/raspberrypi-stable/f/fr24feed/fr24feed_1.0.23-7_armhf.deb
pool/raspberrypi-stable/f/fr24feed/fr24feed_1.0.23-8_armhf.deb
pool/raspberrypi-stable/f/fr24feed/fr24feed_1.0.24-2_armhf.deb
pool/raspberrypi-stable/f/fr24feed/fr24feed_1.0.24-4_armhf.deb
[... 30 lines cut ...]
rpi_binaries/fr24feed_1.0.19-4_armhf.deb
rpi_binaries/fr24feed_1.0.19-5_armhf.deb
rpi_binaries/fr24feed_1.0.19-6_armhf.deb
rpi_binaries/fr24feed_1.0.19-7_armhf.deb
rpi_binaries/fr24feed_1.0.19-8_armhf.deb
rpi_binaries/fr24feed_1.0.19-9_armhf.deb
rpi_binaries/fr24feed_1.0.23-5_armhf.deb
rpi_binaries/fr24feed_1.0.23-7_armhf.deb
rpi_binaries/fr24feed_1.0.23-8_armhf.deb
rpi_binaries/fr24feed_1.0.24-1_armhf.deb
rpi_binaries/fr24feed_1.0.24-2_armhf.deb
rpi_binaries/fr24feed_1.0.24-3_armhf.deb
rpi_binaries/fr24feed_1.0.24-4_armhf.deb
rpi_binaries/fr24feed_1.0.24-5_armhf.deb
rpi_binaries/fr24feed_1.0.24-6_armhf.deb
rpi_binaries/fr24feed_1.0.24-7_armhf.deb
rpi_binaries/fr24feed_1.0.24-8_armhf.deb
rpi_binaries/fr24feed_1.0.25-1_armhf.deb
rpi_binaries/fr24feed_1.0.25-3_armhf.deb
rpi_binaries/fr24feed_1.0.26-4_armhf.deb
rpi_binaries/fr24feed_1.0.26-5_armhf.deb
rpi_binaries/fr24feed_1.0.26-7_armhf.deb
rpi_binaries/fr24feed_1.0.26-8_armhf.deb
rpi_binaries/fr24feed_1.0.26-9_armhf.deb
```

CalvinSchwartz, 2021-01-18:
Sure, Output is
`armv6l`
```
DEB_HOST_ARCH=armhf
DEB_HOST_ARCH_ABI=eabihf
DEB_HOST_ARCH_BITS=32
DEB_HOST_ARCH_CPU=arm
DEB_HOST_ARCH_ENDIAN=little
DEB_HOST_ARCH_LIBC=gnu
DEB_HOST_ARCH_OS=linux
```
And even though it may be not officially supported it definitely works on the Pi Zero.

I checked your updated code and it again fails to detect the architecture when building on the Pi Zero, seems like the variables BUILDPLATFORM and TARGETPLATFORM are empty. I guess I have to set them with the `docker build .` command? (I'm basically just a enduser and do not know exactly how docker build and so on works).
```========== Deploying fr24feed ==========
+ echo 'Running on , building for '
+ case "$TARGETPLATFORM" in
+ echo 'ERROR: Unable to determine architecture or unsupported architecture!'
```

mikenye, 2021-01-18:
Hi @CalvinSchwartz,

That’s interesting. I didn’t realise there were some ARMv6 variants with hardware floating point support. 

Can you try pulling and running `mikenye/fr24feed:latest_arm_v7`? That image is built for `armhf` and should work for you without needing to build yourself.

With regards to your build failing, you’ll need to install [docker buildx](https://docs.docker.com/buildx/working-with-buildx/). That will ensure the missing arguments are set during build time. 

Out of interest, what version of docker are you running? Some research into this issue turned up this: https://github.com/moby/moby/issues/41017. 

It looks like docker can sometimes pick the wrong architecture when an ARMv6 CPU with hardware floating point support is used. In this instance it should likely pick the linux/arm/v7 image... It looks like they may have fixed it in docker v20.10.2...

CalvinSchwartz, 2021-01-18:
Hi,
I tried the arm_v7 but it still fails to start, same error message as before [mikenyefr24feedlatest.log](https://github.com/mikenye/docker-flightradar24/files/5826247/mikenyefr24feedlatest.log).

Docker version is 20.10.2 and I can confirm that I definitely get wrong architecture warnings while building the image (see: [build.log](https://github.com/mikenye/docker-flightradar24/files/5826258/build.log))

With buildx the output is:
`echo 'Running on linux/arm/v6, building for linux/arm/v6'`
Could you then add the arm/v6case to deploy_fr24feed.sh? And is it possible to simply deploy an armv6 image to Docker Hub using GitHub actions? (Never used any of this before, therefore I have no idea how much work that would be.)

mikenye, 2021-01-19:
Hi @CalvinSchwartz,

From your build log it looks like you aren't building against the most recent source - you might need to run a `git pull` on your local repo.

I've added `linux/arm/v6` to the `deploy_fr24feed.sh` script. I'm not sure this is going to work, as the Debian armv6 image uses the `armel` architecture (as opposed to the `armhf` architecture)... Try `mikenye/fr24feed:latest_arm_v6` and let me know how you go.

Regardless, this should hopefully allow you to attempt to build the image yourself.

mikenye, 2021-01-20:
How did you go @CalvinSchwartz ?

CalvinSchwartz, 2021-01-26:
Sorry for the late reply. Your arm v6 container starts without any issues just as it should 👍

mikenye, 2021-01-27:
Thanks for the update.

closed 2021-01-27 by mikenye