sdr-enthusiasts/docker-flightradar24 #26: Beasthost what value should be used?

yousaf465, 2021-06-21:
what value I should use in beasthost? I am trying to install it under docker while running Home assistant on a Rpi 4.

yousaf465, 2021-06-21:
Here is my log file 
`[s6-init] making user provided files available at /var/run/s6/etc...exited 0.,
[s6-init] ensuring user provided files have correct perms...exited 0.,
[fix-attrs.d] applying ownership & permissions fixes...,
[fix-attrs.d] done.,
[cont-init.d] executing container initialization scripts...,
[cont-init.d] 01-fr24feed: executing... ,
[cont-init.d] 01-fr24feed: exited 0.,
[cont-init.d] done.,
[services.d] starting services,
[services.d] done.,
2021-06-21 07:44:35 | ______  _  _         _      _                    _              _____    ___ ,
2021-06-21 07:44:35 | |  ___|| |(_)       | |    | |                  | |            / __  \  /   |,
2021-06-21 07:44:35 | | |_   | | _   __ _ | |__  | |_  _ __  __ _   __| |  __ _  _ __`' / /' / /| |,
2021-06-21 07:44:35 | |  _|  | || | / _` || '_ \ | __|| '__|/ _` | / _` | / _` || '__| / /  / /_| |,
2021-06-21 07:44:35 | | |    | || || (_| || | | || |_ | |  | (_| || (_| || (_| || |  ./ /___\___  |,
2021-06-21 07:44:35 | \_|    |_||_| \__, ||_| |_| \__||_|   \__,_| \__,_| \__,_||_|  \_____/    |_/,
2021-06-21 07:44:35 |                __/ |                                                         ,
2021-06-21 07:44:35 |               |___/                                                          ,
2021-06-21 07:44:35 | info | [httpd]Server started, listening on 0.0.0.0:8754,
2021-06-21 07:44:37 | [i]PacketSenderConfiguration::fetch_config(): Yoda configuration for this receiver is disabled,
2021-06-21 07:44:37 | [d]TLSConnection::ctor(): Enable verify_peer in production code!,
2021-06-21 07:44:37 | [reader][e]Could not connect to tcp://beasthost:30005,
2021-06-21 07:44:38 | [feed][d]fetching configuration,
2021-06-21 07:44:40 | [feed][c]Max range AIR: 350.0nm,
2021-06-21 07:44:40 | [feed][c]Max range GND: 100.0nm,
2021-06-21 07:44:40 | [feed][c]Timestamps: optional,
2021-06-21 07:44:40 | info | Stopping ReceiverACSender threads for feed ,
2021-06-21 07:44:40 | info | Configured ReceiverACSender: 185.218.24.22:8099,185.218.24.23:8099,185.218.24.24:8099, feed: OPLA6, send_interval: 5s, max age: 15s, send metadata: true, mode: 1, filtering: true,
2021-06-21 07:44:40 | info | Network thread connecting to 185.218.24.22:8099 for feed OPLA6,
2021-06-21 07:44:40 | info | [stats]Stats thread started,
2021-06-21 07:44:42 | [reader][e]Could not connect to tcp://beasthost:30005,
2021-06-21 07:44:47 | [reader][e]Could not connect to tcp://beasthost:30005,
2021-06-21 07:44:52 | [reader][e]Could not connect to tcp://beasthost:30005,
2021-06-21 07:44:57 | [reader][e]Could not connect to tcp://beasthost:30005,
2021-06-21 07:45:02 | [reader][e]Could not connect to tcp://beasthost:30005,
2021-06-21 07:45:07 | [reader][e]Could not connect to tcp://beasthost:30005,
2021-06-21 07:45:12 | [reader][e]Could not connect to tcp://beasthost:30005,`

visibilityspots, 2021-06-21:
Hi @yousaf465 this service consumes another one called the beasthost, you could use for example https://github.com/mikenye/docker-readsb-protobuf service. Once this service is running and you have it's ip address you could use that IP as beasthost for the flightradar24 service.

you could combine the docker-compose setup of the readsb-protobuf service -> https://github.com/mikenye/docker-readsb-protobuf#up-and-running-with-docker-compose
with the one of docker-flightradar24 -> https://github.com/mikenye/docker-flightradar24#up-and-running-with-docker-compose
to get them running on the same device.

closed 2021-08-23 by mikenye