sdr-enthusiasts/docker-flightradar24 #38: fr24feed turned off by FR24

puh2oskier, 2021-12-21:
I received an email from FR24 after setting up my raspberrypi 4 adsb setup using https://mikenye.gitbook.io/ads-b/ as a guide.  I have no problems with any of the other accounts I am feeding but FR24 said that my feed was sending a lot of invalid data and turned off my feed.  They said I needed to update to the latest before they would let me try contributing again.  Is there any information I could provide to help me work through this issue.  Even though they turned my feed off I can still see from the mypi address and port that my fr24feed is seeing aircraft.  So I'm not sure what they mean by my feed is sending a lot of invalid data.  Any help would be appreciated.
Thanks

HrJosch, 2021-12-22:
I had set up my Pi4 via this tutorial as well (awesome work btw!) and do experience problems as well with my fr24 container. 
The autoheal container is regularly restarting fr24 docker container and from what I can see inside the fr24 container logs, there seems to be a problem with MLAT timestamps, although the container had been updated by watchtower several times in the last few days.  

`2021-12-22 13:41:51 | info | Network thread connecting to 185.218.24.22:8099 for feed XXXXXXX
2021-12-22 13:42:05 | [mlat][e]Received MLAT timestamp error: 0 seconds!
2021-12-22 13:45:06 | [mlat][e]Received MLAT timestamp error: 0 seconds!
2021-12-22 13:47:17 | [mlat][e]Received MLAT timestamp error: 0 seconds!
2021-12-22 13:50:05 | [mlat][e]Received MLAT timestamp error: 0 seconds!
2021-12-22 13:51:03 | [mlat][e]Received MLAT timestamp error: 0 seconds!
2021-12-22 13:51:51 | info | [stats]sent 1206 bytes
2021-12-22 13:52:48 | [mlat][e]Received MLAT timestamp error: 0 seconds!
2021-12-22 13:55:03 | [mlat][e]Received MLAT timestamp error: 0 seconds!
2021-12-22 13:58:05 | [mlat][e]Received MLAT timestamp error: 0 seconds!
2021-12-22 14:00:47 | [mlat][e]Received MLAT timestamp error: 0 seconds!
2021-12-22 14:01:51 | info | [stats]sent 609 bytes
2021-12-22 14:02:06 | [mlat][e]Received MLAT timestamp error: 0 seconds!
2021-12-22 14:03:32 | [mlat][e]Received MLAT timestamp error: 0 seconds!
2021-12-22 14:04:47 | [mlat][e]Received MLAT timestamp error: 0 seconds!
2021-12-22 14:06:05 | [mlat][e]Received MLAT timestamp error: 0 seconds!
2021-12-22 14:07:49 | [mlat][e]Received MLAT timestamp error: 0 seconds!
2021-12-22 14:09:51 | [mlat][e]Received MLAT timestamp error: 0 seconds!
2021-12-22 14:10:48 | [mlat][e]Received MLAT timestamp error: 0 seconds!
2021-12-22 14:11:52 | info | [stats]sent 1789 bytes
2021-12-22 14:14:03 | [mlat][e]Received MLAT timestamp error: 0 seconds!
2021-12-22 14:15:50 | [mlat][e]Received MLAT timestamp error: 0 seconds!
2021-12-22 14:17:02 | [mlat][e]Received MLAT timestamp error: 0 seconds!
2021-12-22 14:18:47 | [mlat][e]Received MLAT timestamp error: 0 seconds!
2021-12-22 14:20:34 | [mlat][e]Received MLAT timestamp error: 0 seconds!
2021-12-22 14:21:52 | info | [stats]sent 733 bytes
2021-12-22 14:22:04 | [mlat][e]Received MLAT timestamp error: 0 seconds!
2021-12-22 14:23:02 | [mlat][e]Received MLAT timestamp error: 0 seconds!
`

If I can somehow contribute to sort this out, I am willing to do so. :)

puh2oskier, 2021-12-22:
I'm feeling kind of stuck at this point and I'm about a week out from loosing my FR24FEED license and having to start it over because I haven't had time to trouble shoot and when I have I have been unsuccessful at figuring it out on my own. 
I can see that my feed is seeing aircraft, but the system after being up for 18 hours shows the fr24feed container as unhealthy and only up the last 2 hours and then just restarted again, while the rest of the other 6 feeders are operating just great.
I turned off MLAT hoping that would help.  Should fr24feeder on a raspberry pi 4 be running armel/1.0.25-3  I have no idea but I thought it was supposed to be on armhf, unfortunately I've read way to many posts and often times they are outdated so I'm not sure if what I'm looking at applies to my situation.  Any help would be appreciated!
[image]
[image]
[image]

4fb730db30fd   mikenye/adsbexchange:latest      "/init"                  19 hours ago   Up 19 hours (healthy)                                                                                                                    adsbx
043743c9a824   mikenye/piaware:latest           "/init"                  19 hours ago   Up 19 hours (healthy)              80/tcp, 30003/tcp, 30005/tcp, 30105/tcp, 30978-30979/tcp, 0.0.0.0:8081->8080/tcp, :::8081->8080/tcp   piaware
828de9c21358   mikenye/fr24feed:latest          "/init"                  19 hours ago   Up 32 seconds (health: starting)   30003/tcp, 0.0.0.0:8754->8754/tcp, :::8754->8754/tcp, 30334/tcp                                       fr24
9050056075b4   mikenye/dump978:latest           "/init"                  19 hours ago   Up 19 hours (healthy)              30978-30979/tcp, 37981/tcp                                                                            dump978
e57a565dfb4f   mikenye/readsb-protobuf:latest   "/init"                  19 hours ago   Up 19 hours (healthy)              0.0.0.0:8080->8080/tcp, :::8080->8080/tcp                                                             readsb
ed3b7a0cbf7e   mikenye/opensky-network:latest   "/init"                  19 hours ago   Up 19 hours (healthy)                                                                                                                    opensky
77ba9a68cb98   mikenye/tar1090:latest           "/init"                  19 hours ago   Up 19 hours (healthy)              0.0.0.0:8082->80/tcp, :::8082->80/tcp                                                                 tar1090
d4e138a664d4   mikenye/radarbox:latest          "/init"                  43 hours ago   Up 43 hours (healthy)              30105/tcp, 32088/tcp                                                                                  rbfeeder
154182b5e87a   mikenye/planefinder:latest       "/init"                  43 hours ago   Up 43 hours (healthy)              0.0.0.0:30053->30053/tcp, :::30053->30053/tcp, 30054/tcp                                              pfclient
c9a9274a9888   mikenye/adsbhub:latest           "/init"                  43 hours ago   Up 43 hours (healthy)                                                                                                                    adsbhub
c4ce488d1b69   containrrr/watchtower:latest     "/watchtower"            43 hours ago   Up 43 hours                        8080/tcp                                                                                              watchtower
369010a2111b   willfarrell/autoheal:latest      "/docker-entrypoint …"   43 hours ago   Restarting (28) 36 seconds ago                                                                                                           autoheal

HrJosch, 2021-12-23:
@puh2oskier It looks like your fr24 container is also in status starting, just as mine. I am adding screenshots from portainer to make this more clear:
[image]

This is what's happening inside my fr24 container:

[image]

Although it is in status starting, stats are being fed to flightradar24.com and it sees my feeder as online. 

I also discovered there are a lot of TCP socket errors inside the readsb container:

[image]

HrJosch, 2022-01-10:
In the meantime all related containers have been updated several times and my issue described above is completely solved. Everything is running smoothly!  Thank you very much, @mikenye!

mikenye, 2022-01-11:
@HrJosch are you still having issues? I've not been able to replicate your problem. It might be a good idea to join our collective Discord so we can help you troubleshoot.

HrJosch, 2022-01-11:
> @HrJosch are you still having issues? I've not been able to replicate your problem. It might be a good idea to join our collective Discord so we can help you troubleshoot.

Mine is really solved and everything is looking good. Thank you!  I don't know about @puh2oskier.

mikenye, 2022-02-03:
Can anyone affected by this please pull and run the latest version of the image (released just now). Please let me know if that fixes the problem.

puh2oskier, 2022-02-06:
> Can anyone affected by this please pull and run the latest version of the image (released just now). Please let me know if that fixes the problem.

My account was already closed by FlightRadar24 staff because they said my feed was still sending bad data.  I think I may have had some sort of issue with readsb.  If I did docker compose down and then docker compose up... it would scroll and scroll and scroll and readsb would be on the left side and on the right it kept listing not connecting.  Unfortunately I only know enough follow instructions when it comes to linux, not enough to trouble shoot on my own.  I'm going to give it a go using portainer this time on my pi 4 with the 64bit lite version of Rapbian and docker docker on that.  If I run into issues I will look for you over on discord.  Thank You

closed 2022-02-06 by puh2oskier