sdr-enthusiasts/docker-flightradar24 #53: Timestamp errors are back

HrJosch, 2022-02-24:
Since this morning MLAT Timestamp errors are back. The fr24 container remains in status 'starting'. Could you please look into it?

mikenye, 2022-02-24:
@HrJosch can you please post your container log from start up? Please make sure you remove any personally identifiable information before submitting. Thanks.

mikenye, 2022-02-24:
I will note that with the geopolitical climate right now, it looks like FR24 are being pummelled.... :(

HrJosch, 2022-02-24:
[_fr24_logs.txt](https://github.com/sdr-enthusiasts/docker-flightradar24/files/8130247/_fr24_logs.txt)
Sure, logs attached. Thank you!

mikenye, 2022-02-24:
Can you please run `ntpq -pn` on your host, and paste the output?

HrJosch, 2022-02-24:
-bash: ntpq: command not found

mikenye, 2022-02-24:
How are you synchronising the time on your host? I just want to make sure the time on the host is accurate. Typically this is done with NTP or Chrony. It looks like your host does not have NTP installed.

HrJosch, 2022-02-24:
My adsb stack runs on OMV6. Timesync is done via NTP pool.ntp.org

HrJosch, 2022-02-24:
Basically this issue is the same as I described in issue #38 
`[readsb] 2022/02/24 07:15:29 Basestation TCP output: Send Error: Broken pipe: xxx.xx.x.x port 41610 (fd 19, SendQ 88, RecvQ 0)
`

mikenye, 2022-02-24:
What's your source of BEAST data?

Incidentally, I see on their forums:

```
FR24 WILL ONLY USE MLAT DATA CAPTURED WITH 'DVBT'+DUMP1090-MUTABILITY WITH --MLAT OPTION DIRECT (TCP not valid)
```

mikenye, 2022-02-24:
Is your MLAT sync working for other services? It might be worth checking your sync on ADSB Exchange, as they provide pretty good detail about MLAT status.

That will be able to confirm whether the problem is with FR24 or not.

HrJosch, 2022-02-24:
ADSB Exchange looks good:

Feeder ShortID : XXXXX-XXXXX

[MSG-1] ADSBx is recieving your stats data.
This is a separate service from the feeder software.

Aircraft count logged 1600 times (every 60 seconds), last seen (1645686576.4562 seconds since epoc) Thursday 24th of February 2022 07:09:36 AM UTC

HrJosch, 2022-02-24:
Flightaware:

Feeder Mode: Mode-S (1090 MHz)
Multilateration (MLAT): Supported / Enabled

mikenye, 2022-02-24:
See if you can find your ADSBx feeder in here: https://adsbx.org/sync/

mikenye, 2022-02-24:
if you could post a screenshot showing health, timeout, peers, sync-statuses cells for your feeder, that would be great.

HrJosch, 2022-02-24:
[image]
[image]

HrJosch, 2022-02-24:
[image]
[image]

mikenye, 2022-02-24:
It looks like there's nothing wrong with your system. This may be an FR24 server-side issue. Maybe consider disabling MLAT, or see if it eventually comes good.

HrJosch, 2022-02-24:
Meanwhile the FR24 container became green again and says healthy in Portainer.  I noticed that the adsbexchange health status for my feed is unavailable from time to time.

HrJosch, 2022-02-24:
> I will note that with the geopolitical climate right now, it looks like FR24 are being pummelled.... :(

You are absolutely right, they can't handle the requests anymore and established a waiting room as a temporary measure while adding resources.  Thank you for your efforts @mikenye! Keep up the good work!

closed 2023-02-09 by mikenye