sdr-enthusiasts/docker-flightradar24 #58: Allow specifying `bind-interface` via ENV variable

mnkhouri, 2022-05-06:
## Request (TL;DR)

Allow setting `bind-interface` in the [config file](https://github.com/sdr-enthusiasts/docker-flightradar24/blob/a26b5f60a53a386c7717e51b9f368aa13f2663e7/rootfs/etc/cont-init.d/01-fr24feed#L37) via an environment variable, e.g. something like this:

```diff
  # Set up fr24feed
  {
    echo receiver="beast-tcp"
    echo fr24key="${FR24KEY}"
    echo host="${BEASTHOST}:${BEASTPORT}"
    echo bs="no"
    echo raw="no"
    echo logmode="1"
    echo logpath="/var/log"
    echo mlat="${MLAT}"
    echo mlat-without-gps="${MLAT}"
+   if [ -z "${BIND_INTERFACE}" ]; then
+     echo bind-interface="${BIND_INTERFACE}"
+   fi
  } > /etc/fr24feed.ini
```

## Why?

By default, the fr24feed program only allows access to its web interface from IP addresses in a [private range](https://www.okta.com/identity-101/understanding-private-ip-ranges/). If you try to access the web interface from an IP address outside one of these ranges, the following message is displayed:

> For security reasons the web interface is only availble from private class networks or after you have manually specified the bind-interface setting in /etc/fr24feed.ini
> Please set it to bind-interface="0.0.0.0" to accept traffic from all interfaces or to the IP address of your preferred network interface!
> For further assistance please contact support@fr24.com

So why doesn't this affect all users of this container? By default, docker runs with a "userland proxy". This means any incoming connections to a container appear to come from the proxy's address, typically `172.16.0.x` (which is in the private range). So when running this container, most users' connections to the web-ui appears to come from an IP in the private range.

However, I run docker without the userland proxy, so that my containers can [see the real client IPs](https://github.com/docker/for-linux/issues/1309#issuecomment-940386426). I also access my container via the Tailscale VPN, which uses IP addresses _outside_ the private range [ref](https://tailscale.com/kb/1015/100.x-addresses/). So for my particular use case, I need to set the `bind-interface` variable in the config.

Thanks!

mikenye, 2022-05-06:
Thanks for such a detailed issue. I'm happy to implement this. Would you like to submit a PR (so you're listed as a contributor), or would you like me to implement it?

optiz0r, 2022-07-16:
This is also needed for podman, which does not use the proxy either. The proposed fix is not quite right, it should be `-n` instead of `-z` to check if the env var has a value, rather than that it does not. I've done a quick test myself by bind-mounting a modified `cont-init.d/01-fr24feed` startup script, and it appears to work as expected. I'll raise it as a PR.

PR #73 (2022-07-17): Add BIND_INTERFACE env var
Allows setting the `bind-interface` config option in fr24feed.ini
to allow access from non-private IP addresses. Can be set to
`0.0.0.0` to allow access from any IP address (warning, may be
insecure). If env var is not set (default), the config option
will not be added to the config file, which is a secure default.

Fixes #58
commit 7938566f (2022-07-17): Add BIND_INTERFACE env var (#73)

Allows setting the `bind-interface` config option in fr24feed.ini
to allow access from non-private IP addresses. Can be set to
`0.0.0.0` to allow access from any IP address (warning, may be
insecure). If env var is not set (default), the config option
will not be added to the config file, which is a secure default.

Fixes #58
closed 2022-07-17 by mikenye