sdr-enthusiasts/docker-flightradar24 #64: MLAT and multiple network feeds

UhtredTheBold, 2022-05-25:
Hi, I have received an email out of the blue today from flightradar24 asking me to turn off mlat...

> Important information
If you intend to share data to networks alongside Flightradar24, in your Flightradar24 receiver please disable MLAT to the following settings: MLAT=“no”and MLAT-without-gps=“no”. This is to ensure the quality of the data we receive and use and to reduce incompatibility with other services.

I do feed into piaware and flightradar using sdr-enthusiasts containers so I suppose this is what they are referring to.

I don't know if this is just directed at me or if it's going out to a wider audience but in any case I am comfortable I can satisfy their first ask using the documented environment variable:

`MLAT=“no”`

but I don't see an environment variable for the second

`MLAT-without-gps=“no”`

Can an environment variable be added for this and are there any wider consequences from this email that should be added to the container documentation?

mikenye, 2022-05-25:
It’s not directed at you. I and many others in our Discord got the same email. I’ll address this shortly.

mikenye, 2022-05-26:
Hi @UhtredTheBold, if you update your FR24 container, MLAT functionality is completely disabled. Thanks.

closed 2022-05-26 by mikenye

UhtredTheBold, 2022-05-26:
Done. Thank you

PR #65 (2022-05-26): Remove MLAT functionality
Remove all MLAT functionality after many users received an email from FR24 asking them to do so.

[image]

Resolves #64.