sdr-enthusiasts/docker-flightradar24 #68: qemu-arm-static high CPU usage on amd64

rknightion, 2022-06-08:
Hi guys.

Been using this container image for a while now, but recently noticed that the qemu-arm-static fr24 process is using a lot of CPU. Far more than any other feeder/readsb. 
This is on an intel nuc so is amd64 architecture (and it's running the amd64 container arch), but it seems the fr24 binary is being run in an emulated arm mode. 

Is it expected that the fr24 binary is run using arm emulation?

darkorb, 2022-06-23:
Hi folks,

I'm also seeing this - hadn't noticed it before:
[image]

Looks like the repo is set to only install the ARM versions - so seems intended 🤔
https://github.com/sdr-enthusiasts/docker-flightradar24/commit/27a0ae079eeaf6be06408f16997e3f1801ee204c#diff-dd2c0eb6ea5cfc6c4bd4eac30934e2d5746747af48fef6da689e85b752f39557R41

darkorb, 2022-07-29:
@rknightion I just noticed this high CPU usage appears to have stopped happening with a pull I did just now 🤷‍♂️ Do you happen to see the same thing? It's still running under `qemu-arm-static`  but at 0-1% CPU on my box for the most part.

scott-scott, 2022-10-12:
I too am suffering from 100% CPU from qemu-arm-static. I noticed the [run script](https://github.com/sdr-enthusiasts/docker-flightradar24/blob/main/rootfs/etc/services.d/fr24feed/run) tests if it can launch fr24feed natively, so I mounted a freshly downloaded fr24feed_i386 as /usr/local/bin/fr24feed in the container, which launched and work, but now I see fr24feed taking 100% CPU. I guess all I've been able to prove is that it isn't qemu at fault, something else is going on.

mikenye, 2022-12-17:
Hi all, Is this still an issue?

scott-scott, 2022-12-27:
It seems to be the same for me using fr24feed:latest, pulled 5 minute ago. Within around 10 seconds of bringing up the container, qemu-arm-static shows 100% CPU in top. If I disable fr24feed, and run only readsb and rbfeeder, I can see qemu-arm-static (for rbfeeder) using only a tiny amount CPU. I guess this shows my host machine is fine with qemu, just not with fr24feed. Not sure what other information would be helpful for troubleshooting.

QuiGonLeong, 2023-01-13:
Also seeing high CPU utilization with docker-flightradar24:latest - 1.0.30-3. 
I switched from mikenye/fr24feed:1.0.25-3 which didn't have seem to have the high CPU problem.

mikenye, 2023-01-13:
Although there are native binaried for fr24feed, they are not all in version sync. For example, right now on their website, the raspberry pi version is 1.0.30-3, while the amd64 & i386 versions are 1.0.25-3.

Given this, I can't really have a container with mixed version feeders...

Accordingly, I've got the arm version running via qemu in the container. This way everyone is on the same version...

Perhaps what I should do is make architecture specific versions that don't use qemu...

It would be good if fr24 could simply release the same version on all architectures at the same time...

QuiGonLeong, 2023-01-21:
If there is a x64 specific version then it wouldn't have to use qemu emulation which is probably causing the high CPU usage

scott-scott, 2023-01-21:
As noted in my [previous comment](https://github.com/sdr-enthusiasts/docker-flightradar24/issues/68#issuecomment-1275501675), I tried the x86 version in this docker and found the same 100% CPU usage without qemu running.

mikenye, 2023-01-23:
As @scott-scott alludes to, I believe the problem here is with the fr24feed binary. I would suggest reverting to the previous version until FR update their binary (and hopefully fix the issue).

closed 2023-02-09 by mikenye