sdr-enthusiasts/docker-flightradar24 #71: signup.sh: Step 4.1 - Value 7 is incorrect

jtbrough, 2022-07-04:
It seems that the valid values for step 4.1 may have changed:

```
Step 4.1 - Receiver selection (in order to run MLAT please use DVB-T stick with dump1090 utility bundled with fr24feed):

 1 - DVBT Stick (USB)
 -----------------------------------------------------
 2 - SBS1/SBS1er (USB/Network)
 3 - SBS3 (USB/Network)
 4 - ModeS Beast (USB/Network)
 5 - AVR Compatible (DVBT over network, etc)
 6 - microADSB (USB/Network)

Enter your receiver type (1-7)$:7
Value 7 is incorrect, please choose correct receiver type!
```

mikenye, 2022-07-05:
Hi @jtbrough,

Thanks for letting me know. Can you try option 1 and let me know if it works?

wiedehopf, 2022-07-10:
Possible fix, replace this:
```
        echo 'expect "Enter your receiver type (1-7)$:"'
        echo "send \"7\r\""
```
with this (untested):
```
        echo 'expect "Enter your receiver type (1-7)$:"'
        echo "send \"4\r\""
        echo 'expect "Enter your connection type (1-2)$:"'
        echo "send \"1\r\""
        echo 'expect "$:"'
        echo "send \"127.0.0.1\r\""
        echo 'expect "$:"'
        echo "send \"30005\r\""
        echo 'expect "Step 5.1 - Would you like to enable RAW data feed on port 30334 (yes/no)$:"'
        echo "send \"no\r\""
        echo 'expect "Step 5.2 - Would you like to enable Basestation data feed on port 30003 (yes/no)$:
"'
        echo "send \"no\r\""

```

wiedehopf, 2022-07-10:
in this file: https://github.com/sdr-enthusiasts/docker-flightradar24/blob/main/rootfs/scripts/signup.sh#L58

mikenye, 2022-07-11:
I've applied and tested the fix suggested by @wiedehopf in #72.

Once the images finish building (https://github.com/sdr-enthusiasts/docker-flightradar24/actions/runs/2646833967), please pull the latest version of the image and retry the sign-up.

Please re-open this issue if no resolved.

Thanks for your patience.

PR #72 (2022-07-11): Fix issue 71
Fixes #71.
closed 2022-07-11 by mikenye