sdr-enthusiasts/docker-flightradar24 #116: Error: [time][e]Failed to synchronize time

Olivier6767, 2022-12-30:
Since 2 weeks I now have this error every 20 minutes in the FR24FEED container logs.
This does not prevent FR24 to work, my data is uploaded correctly.
However it shows the container as Unhealthy (due to the healthcheck I guess).

I checked my host configuration for time sync and it seems ok:
`timedatectl`
`Local time: Fri 2022-12-30 13:33:15 CET`
`Universal time: Fri 2022-12-30 12:33:15 UTC`
`RTC time: n/a`
`Time zone: Europe/Paris (CET, +0100)`
`System clock synchronized: yes`
`NTP service: active`
`RTC in local TZ: no`

The host is a Raspberry 4 running Ubuntu 22.04 ARM8.

Is this an issue with my setup?
Should the healthcheck be modified?

Johnex, 2022-12-30:
So i did some testing, and came to the conclusion that you might have added this line for the fr24 container to your docker compose file:

      - TZ=${FEEDER_TZ}

Is that correct?

When i added that line, i would have the same issues as you. You could have also seen the message in the logs like so warning about this:

WARNING: Setting timezone via TZ is not supported in this container. fr24feed requires the container has a timezone of GMT (+0).

Some other containers don't like the TZ option as well, such as shipxplorer where the timezone needs to be set in a different way.

Olivier6767, 2022-12-30:
Yes correct, I had the TZ defined in my compose file.
I removed it and it looks like the error is gone.

Thanks!

closed 2022-12-30 by Olivier6767

Johnex, 2022-12-30:
Just for completeness, i have TZ set for the following containers:

readsb, adsbx, mlathub, piaware, pfclient, rbfeeder, rtlsdirairband, tar1090

shipxplorer has the folllowing instead of TZ:

    volumes:
      - /etc/localtime:/etc/localtime:ro
      - /etc/timezone:/etc/timezone:ro

The rest of the containers have no timezone defined.