sdr-enthusiasts/docker-flightradar24 #129: Connection refused in docker-compose with newest update

toooooobi, 2023-02-17:
Hi Mike,

I am creating my stack with following config:

`version: '2.0'

services:

  piaware:
    image: mikenye/piaware:latest
    tty: true
    container_name: piaware
    mac_address: de:ad:be:ef:13:37
    restart: always
    devices:
      - /dev/bus/usb/001/004:/dev/bus/usb/001/004
    ports:
      - 8080:8080
      - 30005:30005
    environment:
      - TZ="Germany/something"
      - LAT=something
      - LONG=something
    volumes:
      - /var/cache/piaware:/var/cache/piaware

  fr24feed:
    image: mikenye/fr24feed:latest
    tty: true
    container_name: fr24feed
    restart: always
    ports:
      - 8754:8754
    environment:
      - BEASTHOST=piaware
      - FR24KEY=something
      - MLAT=yes

  adsbexchange:
   image: mikenye/adsbexchange
   tty: true
   container_name: adsbx
   restart: always
   environment:
     - BEASTHOST=piaware
     - TZ=Germany/something
     - LAT=something
     - LONG=something
     - ALT=50m
     - SITENAME=something
     - UUID=something
   tmpfs:
     - /run:rw,nosuid,nodev,exec,relatime,size=64M,uid=1000,gid=1000
`

This config was running smoothly and perfect for >1year.

Today I did a docker-compose down + docker-compose pull + docker-compose up to update and since them the two containers adsb and fr24feed can not connect to the piaware anymore with following errors.

fr24feed:
`2023-02-17 09:57:30 | BeastBase::connectTcp(): Unable go connect, error: Connection refused[reader][e]Could not connect to tcp://piaware:30005
2023-02-17 09:57:35 | BeastBase::connectTcp(): Unable go connect, error: Connection refused[reader][e]Could not connect to tcp://piaware:30005
2023-02-17 09:57:40 | BeastBase::connectTcp(): Unable go connect, error: Connection refused[reader][e]Could not connect to tcp://piaware:30005
2023-02-17 09:57:45 | BeastBase::connectTcp(): Unable go connect, error: Connection refused[reader][e]Could not connect to tcp://piaware:30005
2023-02-17 09:57:50 | BeastBase::connectTcp(): Unable go connect, error: Connection refused[reader][e]Could not connect to tcp://piaware:30005
`

adsbexchange:
`[mlat-client] Connection to piaware:30005 lost: [Errno 111] Connection refused
[mlat-client] Reconnecting in 10.7 seconds
[adsbexchange-feed] Beast TCP input: Connection to piaware (172.26.0.3) port 30005 failed: 111 (Connection refused)
[mlat-client] Connection to piaware:30005 lost: [Errno 111] Connection refused
[mlat-client] Reconnecting in 10.9 seconds
[mlat-client] Connection to piaware:30005 lost: [Errno 111] Connection refused
`

Output of docker network inspect:
`[
    {
        "Name": "fr24_default",
        "Id": "86ddfba4a073d88fd30456322fa7a32cf125bab49e060cc3f77bd996ba0d9607",
        "Created": "2023-02-17T09:44:47.822618498Z",
        "Scope": "local",
        "Driver": "bridge",
        "EnableIPv6": false,
        "IPAM": {
            "Driver": "default",
            "Options": null,
            "Config": [
                {
                    "Subnet": "172.26.0.0/16",
                    "Gateway": "172.26.0.1"
                }
            ]
        },
        "Internal": false,
        "Attachable": false,
        "Ingress": false,
        "ConfigFrom": {
            "Network": ""
        },
        "ConfigOnly": false,
        "Containers": {
            "4559c33fdf15c4c68c9e51b86419f04979356a3937f1f8e0a4527762fd92671a": {
                "Name": "adsbx",
                "EndpointID": "2e0486c84dd4cc9516ba68a6cf688ee44159532a90a4b8033cdbd6f0e7190cdd",
                "MacAddress": "02:42:ac:1a:00:02",
                "IPv4Address": "172.26.0.2/16",
                "IPv6Address": ""
            },
            "5086a7941ab0becfba06f30aac84387c238c93a900a875a1c03e3fbde5fe7753": {
                "Name": "fr24feed",
                "EndpointID": "bc18b59013a95704acb5788dbd2dadc68bc56435c004af6c7776dcdd5f834475",
                "MacAddress": "02:42:ac:1a:00:04",
                "IPv4Address": "172.26.0.4/16",
                "IPv6Address": ""
            },
            "9782522b6d6f9ccc7f0843602fc7daf787a49bc8880969c8dbbaa35ef8055d56": {
                "Name": "piaware",
                "EndpointID": "dfb3a405b88357a9dcebf32a85fa3d0cb5ffc4cdd969f9dd8eafefdbbbe8ee38",
                "MacAddress": "de:ad:be:ef:13:37",
                "IPv4Address": "172.26.0.3/16",
                "IPv6Address": ""
            }
        },
        "Options": {},
        "Labels": {
            "com.docker.compose.network": "default",
            "com.docker.compose.project": "fr24",
            "com.docker.compose.version": "2.0.0"
        }
    }
]`

Would be great if you could guide me in the right direction. Thanks.

toooooobi, 2023-02-17:
Seems it is a problem of latest piaware container. I used a very old one and it works fine.

mikenye, 2023-02-18:
It may be that FlightAware have stopped piaware listening on port 30005 for beast connections by default.

I would recommend using the readsb-protobuf container as documented here: https://sdr-enthusiasts.gitbook.io/ads-b/

This is likely to be related to https://github.com/sdr-enthusiasts/docker-piaware/issues/140

kx1t, 2024-02-14:
Old issue -- closing

closed 2024-02-14 by kx1t