sdr-enthusiasts/docker-flightradar24 #133: Data not uploading to Flight Radar

garethprice1977, 2023-02-22:
I have setup up the docker container as per the instructions and it is tracking planes just fine from readsb, it claims data is being uploaded but nothing is arriving at Flightradar24
[image]

Sharing key is set fine and I get offline notifications from fr24 when I stop the container, but no aircraft tracking data is uploaded
[image]

dziban303, 2023-02-22:
Please post your compose file's section for the fr24 feeder and we'll have a look.

aklTN, 2023-02-22:
I'm having the same issue. Log had several remarks about unable to sync time, so I removed the "TZ" line from the container, but that didn't help.

garethprice1977, 2023-02-24:
Thanks @aklTN  I didn't have a TZ line so I added one and no difference (worth a try)

This is the tail of my log and it looks like a timeout on the FR24 side

2023-02-24 17:51:55 | [feed][n]syncing stream: timeout
2023-02-24 17:51:10 | [feed][n]syncing stream async: 1
2023-02-24 17:49:39 | [feed][n]syncing stream: timeout
2023-02-24 17:48:53 | [feed][n]syncing stream async: 1
2023-02-24 17:46:30 | [feed][n]syncing stream result: 1
2023-02-24 17:46:29 | [feed][n]ping 1
2023-02-24 17:45:49 | [feed][n]working
2023-02-24 17:45:49 | [feed][n]connected via UDP (fd 23)
2023-02-24 17:45:49 | [feed][n]connecting
2023-02-24 17:45:49 | [feed][n]LFBN2@185.218.24.22:8099/UDP
2023-02-24 17:45:49 | 23-02-24 17:46:50.539 [I][receiver_ac_sender.cpp:36] Network thread connecting to 185.218.24.22:8099 for feed LFBN2
2023-02-24 17:45:49 | 23-02-24 17:46:50.531 [I][crxstats.cpp:588] [stats]Stats thread started
2023-02-24 17:45:49 | 23-02-24 17:46:50.531 [I][receiver_ac_sender.cpp:96] Configured ReceiverACSender: 185.218.24.22:8099,185.218.24.23:8099,185.218.24.24:8099, feed: LFBN2, send_interval: 5s/1s, max age: 15s, send metadata: true, mode: 1, filtering: true
2023-02-24 17:45:49 | 23-02-24 17:46:50.523 [D][receiver_ac_sender.cpp:141] Stop called on non-running network thread for feed 
2023-02-24 17:45:49 | 23-02-24 17:46:50.523 [I][receiver_ac_sender.cpp:137] Stopping ReceiverACSender threads for feed

Kevin10187, 2023-04-03:
This has started this evening after I restarted my rig from a memory upgrade. All other sites are working expect FR24. The local webpage shows aircraft with them being "uploaded" - showing no data on the main FR24 website.

gdelacc, 2023-07-15:
Same problem here:

`[reader][e]Could not connect to tcp://beasthost:30005`

Checking live tracking with: `docker exec -it readsb viewadsb` and working fine:
[image]

Link with fr24 account is perfect:
[image]

And check container errors: 
[image]

My docker ps:
[image]

What is wrong?

Thanks in advance!

kx1t, 2024-02-14:
This looks like a configuration error by the user. Make sure that `- BEASTHOST` is set to the actual container name of your adsb receiver. If you are using `ultrafeeder`, then it's probably `ultrafeeder`. If you are using `readsb` or `readsb-protobuf`, it's probably `readsb`. Etc.

closed 2024-02-14 by kx1t

chrisbrownie, 2024-04-30:
@garethprice1977 did you get this resolved? 

I'm seeing the same issue now. Data is getting to the FR24 container ("Aircraft Tracked" and "Aircraft Uploaded" both > 0), but FR24 _My Data sharing_ is showing "Online - No data".

pilaGit, 2025-08-28:
Docker should recognize `ultrafeeder` as the container name and pull its IP address. 

The fr24feed container running in Docker on my LibreElec will not work if I try to use `ultrafeeder` as docs say. All is well when i configure fr24feed to use the actual IP address `172.17.0.2` belonging to the `ultrafeeder` container.

It ca be read easily with `docker inspect ultrafeeder` but if the address changes, it will stop working.