sdr-enthusiasts/docker-flightradar24 #224: FR24

billcastroTT, 2024-01-01:
when I run the script in the instruction I get the following error:

s6-envdir: fatal: unable to envdir /run/s6/container_environment: No such file or directory

lg, 2024-01-24:
hey i got this working by just following the manual instructions either: https://sdr-enthusiasts.gitbook.io/ads-b/feeder-containers/feeding-flightradar24#manual-sign-up-method or https://github.com/sdr-enthusiasts/docker-flightradar24#script-method. i forget which one i did which made this work, but one of the two should.

kbob76, 2024-01-26:
Yup manual method worked, had to get around by exec'ing into the container and manually adding the code snippet to the prometheus yaml file.

Guzbikes, 2024-02-07:
I'm also getting the same error. 

But when I used: 
https://github.com/sdr-enthusiasts/docker-flightradar24?tab=readme-ov-file#script-method
and filled in the fields myself, it worked fine

The only difference in https://sdr-enthusiasts.gitbook.io/ads-b/feeder-containers/feeding-flightradar24#automatic-sign-up-script-method is the line: `--entrypoint /scripts/signup.sh`

Interesting, have to do some more poking around. Anyone else figure it out?

kx1t, 2024-02-14:
The signup process was indeed broken after changes made by FR24. As a result, we completely overhauled the signup process in the last few days. Using the new instructions in the README.md, you should be able to sign up for an ADSB feeder key and for a UAT feeder key.

closed 2024-02-14 by kx1t

lg, 2024-02-14:
clever to use qemu to run the damn thing 👏