sdr-enthusiasts/docker-flightradar24 #243: cant view local stats page for UAT

gottaloveit, 2024-02-28:
Similar to the stats page on port 8754, the UAT feeder exposes 8755, but the BIND_INTERFACE is apparently not writing to the UAT feed config as shown by the following error when I add port 8755 to the ports in docker-compose.yml

[image]

the stats page on 8754:
[image]

here's my docker-compose for fr24
```
  fr24:
    # fr24 feeds ADS-B and UAT data (from ultrafeeder) to FlightRadar24. It also includes a status website. Please be careful
    # not to expose the status website to the internet as users may be able to start/stop/change the service from there.
    # Also note that FR24 has requested NOT to enable MLAT for those station that feed to multiple services; as such, it's commented out.
    image: ghcr.io/sdr-enthusiasts/docker-flightradar24
    #    profiles:
    #      - donotstart
    tty: true
    container_name: fr24
    hostname: fr24
    restart: always
    labels:
      - "autoheal=true"
    ports:
      - 8754:8754
      - 8755:8755
    environment:
      - BEASTHOST=ultrafeeder
      - TZ=${FEEDER_TZ}
      - FR24KEY=${FR24_SHARING_KEY}
      - FR24KEY_UAT=${FR24_UAT_SHARING_KEY}
      - BIND_INTERFACE=0.0.0.0
      - UATHOST=dump978
      - UATPORT=30978
    tmpfs:
      - /var/log
```

fredclausen, 2024-03-18:
I have merged a fix for this. Please let me know if this fixes your problem!

gottaloveit, 2024-03-19:
it does .. tyvm

closed 2024-03-19 by gottaloveit