sdr-enthusiasts/docker-flightradar24 #280: Cannot connect to readsb container, failed to synchonize time

harenber, 2024-07-13:
It seems I cannot get the fr24feed to run. 

Here is my `docker-compose.yml` (some personal information replaced with XXX):

```
cubietruck:/container# cat fr24/docker-compose.yml
version: '3.7'

services:
volumes:
  readsbpb_rrd:
  readsbpb_autogain:

services:
  readsb:
    image: ghcr.io/sdr-enthusiasts/docker-readsb-protobuf:latest
    tty: true
    container_name: readsb
    hostname: readsb
    restart: always
    devices:
      - /dev/bus/usb:/dev/bus/usb
    ports:
      - 8080:8080
      - 30005:30005
    environment:
      - TZ="Europe/Berlin"
      - READSB_DCFILTER=true
      - READSB_DEVICE_TYPE=rtlsdr
      - READSB_FIX=true
      - READSB_GAIN=autogain
      - READSB_LAT=XXX
      - READSB_LON=XXX
      - READSB_MODEAC=true
      - READSB_RX_LOCATION_ACCURACY=2
      - READSB_STATS_RANGE=true
      - READSB_NET_ENABLE=true
    volumes:
      - readsbpb_rrd:/run/collectd
      - readsbpb_autogain:/run/autogain
      - /proc/diskstats:/proc/diskstats:ro
    tmpfs:
      - /run/readsb:size=64M
      - /var/log:size=32M

  fr24feed:
    container_name: fr24
    image: mikenye/fr24feed:latest
    tty: true
    ports:
      - 8754:8754
    volumes:
      - /etc/localtime:/etc/localtime:ro
    environment:
      - TZ="Europe/Berlin"
      - FR24KEY=XXX
      - MLAT=no
      - BEASTHOST=readsb
```

Error messages are like:

```
cubietruck:/container# docker logs -f fr24
[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] 00-libsecomp2: executing... 
[cont-init.d] 00-libsecomp2: exited 0.
[cont-init.d] 01-fr24feed: executing... 
WARNING: Setting timezone via TZ is not supported in this container. fr24feed requires the container has a timezone of GMT (+0).
fr24feed version: 1.0.34-0
[cont-init.d] 01-fr24feed: exited 0.
[cont-init.d] 02-show-architecture.sh: executing... 

Hardware information:
Machine:   x86_64
Processor: unknown
Platform:  unknown

[cont-init.d] 02-show-architecture.sh: exited 0.
[cont-init.d] done.
[services.d] starting services
[services.d] done.
2024-07-13 10:54:06 | ______  _  _         _      _                    _              _____    ___ 
2024-07-13 10:54:06 | |  ___|| |(_)       | |    | |                  | |            / __  \  /   |
2024-07-13 10:54:06 | | |_   | | _   __ _ | |__  | |_  _ __  __ _   __| |  __ _  _ __`' / /' / /| |
2024-07-13 10:54:06 | |  _|  | || | / _` || '_ \ | __|| '__|/ _` | / _` | / _` || '__| / /  / /_| |
2024-07-13 10:54:06 | | |    | || || (_| || | | || |_ | |  | (_| || (_| || (_| || |  ./ /___\___  |
2024-07-13 10:54:06 | \_|    |_||_| \__, ||_| |_| \__||_|   \__,_| \__,_| \__,_||_|  \_____/    |_/
2024-07-13 10:54:06 |                __/ |                                                         
2024-07-13 10:54:06 |               |___/                                                          
2024-07-13 10:54:06 | 24-07-13 10:54:06.575 [I][http-server.cpp:270] [httpd]Server started, listening on :::8754
2024-07-13 10:54:06 | [i]PacketSenderConfiguration::fetch_config(): Yoda configuration for this receiver is disabled
2024-07-13 10:54:06 | [d]TLSConnection::ctor(): Enable verify_peer in production code!
2024-07-13 10:54:06 | BeastBase::connectTcp(): Unable go connect, error: Cannot assign requested address[reader][e]Could not connect to tcp://readsb:30005
2024-07-13 10:54:07 | [time][e]Failed to synchronize time
```

What I already tested:

- I tried to remove the `TZ` environment as suggested in https://github.com/sdr-enthusiasts/docker-flightradar24/issues/116
- The readsb container seems to run fine. See the data when running the `viewadsb` command.
- I can also telnet from both containers to readsb port 30005 (so a `telnet readsb 30005` shows quite a lot of data).
- I also pointed fr24feed to the hostname of the docker host and also checked that this is connectable:

```
cubietruck:/container# telnet cubietruck.fritz.box 30005
Trying 127.0.0.1...
Connected to cubietruck.fritz.box.
Escape character is '^]'.
3M #X2*`2,3M #
              2x8233M #ܞ2
```

But whatever I do, the fr24feed container is still complaining as mentioned above.

Any pointers would be really appreciated.

Thanks, Torsten

harenber, 2024-07-16:
Thanks for the suggestion @JMTK . Already tried that:

```
cubietruck:/container# ls -l /etc/localtime 
lrwxrwxrwx    1 root     root            17 Jul 16 15:27 /etc/localtime -> /etc/zoneinfo/UTC
cubietruck:/container# date
Tue Jul 16 15:31:23 UTC 2024
cubietruck:/container# docker compose up -d
[+] Running 3/3
 ✔ Network fr24_default  Created                                                                                                                                                                              0.1s 
 ✔ Container readsb      Started                                                                                                                                                                              0.3s 
 ✔ Container fr24        Started                                                                                                                                                                              0.3s 
cubietruck:/container# docker compose logs fr24feed
fr24  | [s6-init] making user provided files available at /var/run/s6/etc...exited 0.
fr24  | [s6-init] ensuring user provided files have correct perms...exited 0.
fr24  | [fix-attrs.d] applying ownership & permissions fixes...
fr24  | [fix-attrs.d] done.
fr24  | [cont-init.d] executing container initialization scripts...
fr24  | [cont-init.d] 00-libsecomp2: executing... 
fr24  | [cont-init.d] 00-libsecomp2: exited 0.
fr24  | [cont-init.d] 01-fr24feed: executing... 
fr24  | WARNING: Setting timezone via TZ is not supported in this container. fr24feed requires the container has a timezone of GMT (+0).
fr24  | fr24feed version: 1.0.34-0
fr24  | [cont-init.d] 01-fr24feed: exited 0.
fr24  | [cont-init.d] 02-show-architecture.sh: executing... 
fr24  | 
fr24  | Hardware information:
fr24  | Machine:   x86_64
fr24  | Processor: unknown
fr24  | Platform:  unknown
fr24  | 
fr24  | [cont-init.d] 02-show-architecture.sh: exited 0.
fr24  | [cont-init.d] done.
fr24  | [services.d] starting services
fr24  | [services.d] done.
fr24  | 2024-07-16 15:31:27 | ______  _  _         _      _                    _              _____    ___ 
fr24  | 2024-07-16 15:31:27 | |  ___|| |(_)       | |    | |                  | |            / __  \  /   |
fr24  | 2024-07-16 15:31:27 | | |_   | | _   __ _ | |__  | |_  _ __  __ _   __| |  __ _  _ __`' / /' / /| |
fr24  | 2024-07-16 15:31:27 | |  _|  | || | / _` || '_ \ | __|| '__|/ _` | / _` | / _` || '__| / /  / /_| |
fr24  | 2024-07-16 15:31:27 | | |    | || || (_| || | | || |_ | |  | (_| || (_| || (_| || |  ./ /___\___  |
fr24  | 2024-07-16 15:31:27 | \_|    |_||_| \__, ||_| |_| \__||_|   \__,_| \__,_| \__,_||_|  \_____/    |_/
fr24  | 2024-07-16 15:31:27 |                __/ |                                                         
fr24  | 2024-07-16 15:31:27 |               |___/                                                          
fr24  | 2024-07-16 15:31:27 | 24-07-16 15:31:27.936 [I][http-server.cpp:270] [httpd]Server started, listening on :::8754
fr24  | 2024-07-16 15:31:28 | [i]PacketSenderConfiguration::fetch_config(): Yoda configuration for this receiver is disabled
fr24  | 2024-07-16 15:31:28 | [d]TLSConnection::ctor(): Enable verify_peer in production code!
fr24  | 2024-07-16 15:31:28 | BeastBase::connectTcp(): Unable go connect, error: Network is unreachable[reader][e]Could not connect to tcp://readsb:30005
fr24  | 2024-07-16 15:31:28 | [time][e]Failed to synchronize time
fr24  | 2024-07-16 15:31:33 | BeastBase::connectTcp(): Unable go connect, error: Network is unreachable[reader][e]Could not connect to tcp://readsb:30005
fr24  | 2024-07-16 15:31:35 | [time][e]Failed to synchronize time
cubietruck:/container# 
```

Still doesn't seem to work. 

When I remove the "TZ" variable from the containers, at least the "WARNING: Setting timezone via TZ is not supported in this container. fr24feed requires the container has a timezone of GMT (+0)." message disappears, but otherwise:

```
cubietruck:/container# docker compose logs fr24feed
fr24  | [s6-init] making user provided files available at /var/run/s6/etc...exited 0.
fr24  | [s6-init] ensuring user provided files have correct perms...exited 0.
fr24  | [fix-attrs.d] applying ownership & permissions fixes...
fr24  | [fix-attrs.d] done.
fr24  | [cont-init.d] executing container initialization scripts...
fr24  | [cont-init.d] 00-libsecomp2: executing... 
fr24  | [cont-init.d] 00-libsecomp2: exited 0.
fr24  | [cont-init.d] 01-fr24feed: executing... 
fr24  | fr24feed version: 1.0.34-0
fr24  | [cont-init.d] 01-fr24feed: exited 0.
fr24  | [cont-init.d] 02-show-architecture.sh: executing... 
[... 71 lines cut ...]
update-alternatives: using /usr/bin/telnet.netkit to provide /usr/bin/telnet (telnet) in auto mode
update-alternatives: warning: skip creation of /usr/share/man/man1/telnet.1.gz because associated file /usr/share/man/man1/telnet.netkit.1.gz (of link group telnet) doesn't exist
root@92e3b04ac2bb:/# telnet readsb 30005
Trying 172.18.0.3...
Connected to readsb.
Escape character is '^]'.
,2(XL1}c01}1d2}>  3},
               ` *2}
j("YS1}
v#Q43}V| M$
           mЀ2}{>
```

`/var/log/fr24feed.log` has these kind of messages:

```
2024-07-16 15:39:26 | [reader][i]Connecting to unknown receiver via (tcp://readsb:30005)
2024-07-16 15:39:26 | BeastBase::connectTcp(): Unable go connect, error: Network is unreachable[reader][e]Could not connect to tcp://readsb:30005
2024-07-16 15:39:26 | [time][i]Synchronizing time via NTP
2024-07-16 15:39:27 | [time][e]Failed to synchronize time
```

Cheers

  Torsten

JMTK, 2024-07-16:
I had a similar issue... 2 things helped:
It turned out the hostname was wrong on mine, but yours looks correct (readsb). Mine was `ultrafeeder` in this case.

Have you tried rebuilding the entire virtual network via `docker compose down`? 

Try adding ```- VERBOSE_LOGGING=true``` to the fr24 container env variables

wiedehopf, 2024-07-28:
First off this is an outdated container version:     image: mikenye/fr24feed:latest

Overall possibly just re-do your stack following this guide: https://sdr-enthusiasts.gitbook.io/ads-b/

Now the changed fr24feed version isn't guaranteed to work but it might.
Leading theory to fr24 issues like this is that it can't handle certain DNS replies when it looks up the ntp servers.
Not sure though.

Really it's just very lackluster software and fr24 provides bad support.

harenber, 2024-07-30:
Thansk @JMTK for the suggestions. Yes, I completely `docker compose down` before re-starting. Didn't help either.

Never mind, I installed the stuff now bare-metal. Seems fr24feed is really not designed in a way that it could be easily moved into a container.

closed 2024-07-30 by harenber

wiedehopf, 2024-07-30:
It works just fine for most people in a container.
When it doesn't work it's very hard and frustrating to track down programming errors in a program that's closed source.

Did you try the current version?
```yml
    image: ghcr.io/sdr-enthusiasts/docker-flightradar24:latest

```
Your yml indicated you were using a very old version from dockerhub.

0tofu, 2024-08-06:
I encountered the same issue.

I am using the image `ghcr.io/sdr-enthusiasts/docker-flightradar24:latest`.
Adding `network_mode: host` to `fr24feed` in `compose.yaml` resolved the problem.
However, since I don't understand the cause, if there is a way to solve it without using `network_mode: host`, please let me know.